wokwi/avr8js #11: Implemenet serial port (USART) RX

urish, 2020-02-03:
USART transmit implemented in #6, we need to implement RX as well!

yiivon, 2020-07-10:
Consider following please, maybe it will help a bit 
```

export class AVRUSART {
  public onByteTransmit: USARTTransmitCallback | null = null;
  public onLineTransmit: USARTLineTransmitCallback | null = null;

  private lineBuffer = '';
  private outBuffer = [];

  constructor(private cpu: CPU, private config: USARTConfig, private freqMHz: number) {
   ...
    this.cpu.readHooks[config.UDR] = (v) => {
      const d: u8 = this.outBuffer.shift();
      return (this.cpu.data[this.config.UDR] = d);
    };
  }

  tick() {
    if (this.cpu.interruptsEnabled) {
      const ucsra = this.cpu.data[this.config.UCSRA];
      const ucsrb = this.cpu.data[this.config.UCSRB];
 ...
      if (ucsra & UCSRA_UDRE && this.outBuffer.length > 0) {
        avrInterrupt(this.cpu, this.config.rxCompleteInterrupt);
      }
    }
  }

  write(b: u8) {
    this.outBuffer.push(b); 
  }

  writeString(s: string) {
    for (let i = 0; i < s.length; i++) {
      this.outBuffer.push(s.charCodeAt(i));
    }
  }

  writeln(s: string) {
    s += '\n';
    this.writeString(s);
  }
 ...
}

```

urish, 2020-07-11:
That is helpful, thank you!

yiivon, 2020-07-23:
In fact, I am very eager for avr8js to run [grbl](https://github.com/gnea/grbl) [firmware](https://github.com/gnea/grbl/releases/download/v1.1h.20190825/grbl_v1.1h.20190825.hex) completely and accurately.

GRBL is a firmware implemented in pure C language running on arduino uno (ATmega328P).

I use the latest version of Proteus to run grbl for arduino simulation, and I can get the correct result. But the current avr8js does not seem to be able to achieve the same, I guess it may be that some implementations of timer are not perfect.

If you also want to make avr8js run like real chips, you can try to run grbl firmware in emulation, which may solve this problem and make avr8js close to real chips again. At the same time, it also helped me a lot.

Ask for your help and thank you sincerely!

urish, 2020-07-23:
Thanks for the feedback @yiivon! Can you please open a new issue for this?

urish, 2020-09-30:
Until this is implemented, here is a quick hackish workaround. The code assumes that you have a list of bytes pending for Serial RX inside the serialBuffer array: 

1. When initializing the simulation, add the following code:

```typescript
cpu.readHooks[usart0Config.UDR] = () => serialBuffer.shift() || 0;
```

2. Then, add this inside your [execute()](https://github.com/wokwi/avr8js/blob/377058623b33e73d3d30714c14080c22d01fd3ab/demo/src/execute.ts#L51) loop:

```typescript
const ucsra = cpu.data[usart0Config.UCSRA];
if (ucsra & 0x20 && serialBuffer.length > 0) {
  avrInterrupt(cpu, usart0Config.rxCompleteInterrupt);
}
```

closed 2021-02-19 by urish

urish, 2021-02-19:
Ok, so this is now released!

Usage example: https://stackblitz.com/edit/avr8js-serial-rx?file=index.ts

commit cf13e06a (2021-02-19): feat(usart): implement RX #11

close #11