wokwi/avr8js #16: Serial port (USART TX) should respect the given baud rate

urish, 2020-03-15:
In the current implementation, whenever a byte is sent through the Serial port (USART), we [immediately acknowledge it](https://github.com/wokwi/avr8js/blob/2baf7d362b7ac0e7890e35287d6043757ee9c77d/src/usart.ts#L91). In practice, sending data through the AVR Serial port takes time. For instance, if we use 9600 baud rate, we can send about 960 bytes per second (each byte requires 10 bits in total: 1 start bit, 8 data bits, and 1 stop bit). Thus, we should only mark the transfer completed after ~1.04ms (or ~16667 clock cycles when running in 16MHz).

In other words, when running the following program:

```cpp
void setup() {
  Serial.begin(9600);
}

void loop() {
  Serial.println("Hello, Serial!");
}
```

We should get about 60 lines printed every second ("Hello, Serial!\r\n" is 16 bytes, and 960/16 == 60).

When calculating the delay before setting the TX Complete flag after each byte transmission, we ideally need to take the following parameters into account:
- The baud rate (we already calculate it in the `baudRate` getter)
- The number of bits per character (see `bitsPerChar` getter)
- The presence of a parity bit (check the datasheet for more info)
- The number of stop bits (1 or 2)

In practice, most implementations only change the baud rate (9600 and 115200 are probably the most common values), and the other parameters are static (8 bits per bytes, no parity, 1 stop bit), so we can decide to start only with the baud rate.

Information about the USART registers can be found in page 200 / section 20.11 of the [datasheet](http://ww1.microchip.com/downloads/en/DeviceDoc/ATmega48A-PA-88A-PA-168A-PA-328-P-DS-DS40002061A.pdf).

LironHazan, 2020-03-19:
:sparkle: on it :)

LironHazan, 2020-03-19:
@urish Thanks!

commit 3d095d34 (2020-11-26): fix(usart): respect the given baud rate #16

close #16
closed 2020-11-26 by urish