wokwi/avr8js #26: Attaching pushbutton input to PIN register

gfeun, 2020-03-31:
Hello again !

So I have made some progress on my virtual training board using avr8js.
I have 8 leds + 8 pushbutton.
Using demo project it was quite easy to set up.

I am now wondering if there is a "nice" way to hook up the pushbuttons to a specific µController GPIO port.

What I am doing now is register some event listener on the push-button events. On event, I access cpu register directly and modify the internal cpu data field of the corresponding register

```typescript
function buttonPress(e :Event) {
  const id = e.target.id.slice(-1)
  const bitMask = 1 << id;
  runner.cpu.data[runner.portC.portConfig.PIN] = runner.cpu.data[runner.portC.portConfig.PIN] | bitMask
}

function buttonRelease(e :Event) {
  const id = e.target.id.slice(-1)
  const bitMask = ~(1 << id);
  runner.cpu.data[runner.portC.portConfig.PIN] = runner.cpu.data[runner.portC.portConfig.PIN] & bitMask
}

// buttons ids are "button1", "button2"
const buttonModule = new Array<PushButtonElement>(8);
for(let i=0; i<buttonModule.length; i++) {
  buttonModule[i] = document.querySelector<PushButtonElement>("wokwi-pushbutton[id=button"+i+"]");
  buttonModule[i].addEventListener("button-press", buttonPress, false)
  buttonModule[i].addEventListener("button-release", buttonRelease, false)
}
```

This is working, with the following code i have a "blink led according to button pressed":
 ```c++
setup() {
  DDRB = 0xFF;
  DDRC = 0x00;
}
loop() {
  delay(500);
  PORTB = PINC;
  delay(500);
  PORTB = 0;
}
```

But i find modifying the internal cpu data directly looks like a hack and i may have missed the "proper" way to do it. Do you have any recommendation or good practice on how to do it ?

Thank you !

urish, 2020-03-31:
Hi Glenn!

Great to hear from you again :-)

At the moment, the [GPIO module](https://github.com/wokwi/avr8js/blob/4e77e3380f313d024d04d42d77f129ab5df72699/src/peripherals/gpio.ts#L133) only includes the `pinState()` function, which queries the current state of the pins.

It does make a lot of sense to add another method that will let you set the state of a pin. One of the main challenges is correctly dealing with pull-ups. For instance, let's say that your virtual button connects the pin with GND. When the button is not pressed, the pin is floating. What value should the user code get when reading from the pin if the pull-up is not used?

urish, 2020-05-10:
We now have a new `setPin()` method for this purpose. Note that this is a breaking change since now setting the `PIN` memory location directly (as you did above) will be overridden by the `AVRIOPort` class.

closed 2020-05-10 by urish

urish, 2020-05-25:
Released as part of 0.9.0

commit eed745ea (2020-05-10): feat(gpio): add setPin() function

close #26