wokwi/avr8js #43: AssemblyScript conflicts

Dudeplayz, 2020-05-03:
This issue collects conflicts of the current code base and AssemblyScript. See #35 and the [research project](https://github.com/wokwi/avr8js-research).

Found problems:
1. Data types must be specified explicitly
2. AssemblyScript has a [restricted data type set](https://docs.assemblyscript.org/basics/types)
   - The current `types` file can be adapted to keep JS compatibility
3. Interfaces are currently not supported. Breaks e.g. `ICPU` and `CPU`. (further research required)
4. Declaring variables as constructor parameters brings up memory access errors because the [instantiation works differently](https://docs.assemblyscript.org/details/memory). E.g. `CPU` with `progMem`, which has to be declared outside the constructor and the following instantiations have to be done in the constructor body:
```typescript
    constructor(progMem: Uint16Array, private sramBytes: u64 = 8192) {
        this.progMem = progMem;
        this.progBytes = Uint8Array.wrap(this.progMem.buffer);
        this.pc22Bits = this.progBytes.length > 0x20000;
        this.reset();
    }
```

(The research is still ongoing.)

MaxGraey, 2020-05-03:
3. Some basic interfaces should be allowed in assemblyscript@nightly with recent [PR](https://github.com/AssemblyScript/assemblyscript/pull/1208)

MaxGraey, 2020-05-09:
4. [fixed](https://github.com/AssemblyScript/assemblyscript/pull/1255)

urish, 2020-05-09:
Thank @MaxGraey for your help here :-) It's really appreciated!

Dudeplayz, 2020-06-25:
@MaxGraey There is still an error if a class variable is based on a constructor variable:
```
RuntimeError: memory access out of bounds
    at ~lib/arraybuffer/ArrayBuffer#get:byteLength (wasm-function[26]:0xfd6)
    at ~lib/typedarray/Uint8Array.wrap (wasm-function[31]:0x11c5)
    at ~lib/typedarray/Uint8Array.wrap@varargs (wasm-function[32]:0x12a6)
    at assembly/cpu/cpu/CPU#constructor (wasm-function[35]:0x1345)
```

I had to change this:
```typescript
readonly progBytes : Uint8Array = Uint8Array.wrap(this.progMem.buffer);
constructor(public progMem: Uint16Array, private sramBytes: u64 = 8192) {
    this.reset();
}
```

To that:
```typescript
readonly progBytes : Uint8Array;
constructor(public progMem: Uint16Array, private sramBytes: u64 = 8192) {
    this.progBytes = Uint8Array.wrap(this.progMem.buffer);
    this.reset();
}
```

MaxGraey, 2020-06-25:
Could you create minimal example on WebAssembly Studio?

Dudeplayz, 2020-06-25:
@MaxGraey

> 1. Some basic interfaces should be allowed in assemblyscript@nightly with recent [PR](https://github.com/AssemblyScript/assemblyscript/pull/1208)
> 2. this is known [issue](https://github.com/AssemblyScript/assemblyscript/issues/1225)

Is the mentioned PR already in 0.12.3? I tried to use the CPU Interface again, but it doesn't work with the error, that the variables are declared twice with the same name.

MaxGraey, 2020-06-25:
Without code I can't tell what's wrong. Plz open issue with minimal reproducible example

Dudeplayz, 2020-06-25:
Ok, no problem.

Dudeplayz, 2020-06-25:
Yes, I will open an issue.

closed 2021-04-10 by urish