wokwi/avr8js #48: The right way to detect when the TX Led is ON or OFF

tawjaw, 2020-06-02:
Hi @urish 
How can I detect when the TX Led is ON or OFF?
My guess is that whenever there's transmission on USART then the LED should be on. 
But how to switch it off when there's nothing being transmitted?

urish, 2020-06-02:
Yes, the recommended method is to listen for `onByteTransmit` callback of the `AVRUSART` instance. Whenever you get a byte, you switch the LED on for the next browser frame. You could use a flag that will be set to true in `onByteTransmit`, then check it and turn the LED on in `requestAnimationFrame` iff that flag is true, and set the flag to false so that next animation frame will turn the LED off (unless there's been another `onByteTransmit` call in the interim).

Ideally, you could use the baud rate (as outlined in #16) to calculate the exact duty cycle of the LED (each byte would be about 10 symbols, and you can derive the time-per-symbol from the baud rate), but I believe for most purposes briefly turning it on whenever there is a new byte would suffice.

closed 2020-06-03 by urish

tawjaw, 2020-07-29:
Thanks @urish I just saw this for some reason!

urish, 2020-07-29:
You're welcome :-)