wokwi/avr8js #49: NodeJS demo example ?!

ricardojlrufino, 2020-06-07:
First, congratulations on the project.
I ended up not finding a pure nodejs version of the examples. and created the repository:
https://github.com/ricardojlrufino/avr8js_demo_node

I am thinking of some way of integrating with an application made in JAVA, a plugin for arduino IDE.
Java supports JavaScript code execution, but I believe it will be slow.
My first attempt was this and it got really slow: using: graalvm / graaljs
https://github.com/graalvm/graaljs/issues/296

urish, 2020-06-07:
Thanks for sharing your demo!
It's interesting to see the difference between Node.js and graalVM. 

How did you find about the AVR8js project?

ricardojlrufino, 2020-06-07:
searching for Arduíno emulator.
would it be complex to port or code to Java?

urish, 2020-06-07:
It'd probably take a few weeks of work. I think a more promising approach would be to call NodeJS (or another JavaScript runtime) from Java. You can probably even use the bare V8 JavaScript engine from Java, I'm pretty sure there are bindings available, and I think the simulator should run there pretty smoothly.

I have recently tried to run it on [deno](https://deno.land/), which is also based on V8, and it worked flawlessly (I hope to get that demo to YouTube at some point).

So basically you found it on Google? I'm wondering how visible the project is for people who are looking for an Arduino emulator.

ricardojlrufino, 2020-06-07:
Yes i'm running using graavm and j2v8.
but both are considerably slower compared to the native nodejs.
I'm running from the eclipse, I'll try to pack a jar and see if anything changes.

```java
public class J2V8 {
  public static void main(String[] args) throws IOException {
    Path path = Paths.get("/media/ricardo/Dados/TEMP/testes-java/avr8js_demo_vanilla/run-v8.js");

    NodeJS node = NodeJS.createNodeJS();
    node.exec(path.toFile());

    while (node.isRunning()) {
      node.handleMessage();
    }
  }
}

```

i need remove async / await to work

ricardojlrufino, 2020-06-07:
> So basically you found it on Google? I'm wondering how visible the project is for people who are looking for an Arduino emulator.

It was actually a video on youtube ..

I am making a number of improvements to the Arduino IDE, and one of the things I wanted to integrate was an emulator.
https://github.com/ricardojlrufino/Arduino

Now i'm building a HMI / Interface builder...
[image]

To make easy create visualizations and experiments..

ricardojlrufino, 2020-06-07:
[image]

urish, 2020-06-07:
That's super-cool Ricardo!

I think that @LironHazan and @vitaliy-bobrov would love to see what you are doing here with the guitar effects :)

I'm not experienced with J2V8, which V8 version does it use? Perhaps @benjamingr can give you some pointers if you get stuck on figuring out the performance issues...

ricardojlrufino, 2020-06-07:
In fact, it runs at about the same speed as the browser ... on nodejs much faster ... I was impressed.

ricardojlrufino, 2020-06-07:
> I'm not experienced with J2V8, which V8 version does it use?
```
    <dependency>
            <groupId>com.eclipsesource.j2v8</groupId>
            <artifactId>j2v8_linux_x86_64</artifactId>
            <version>4.8.0</version>
        </dependency>
```

urish, 2020-06-07:
Which speeds did you observe on Node vs the browser? 

There can be many factors which affect the speed of execution. For example, transpiling the code down to ES5 can slow down execution by adding extra code...

ricardojlrufino, 2020-06-07:
simple benckmark
https://github.com/ricardojlrufino/avr8js_demo_node/tree/vanilla

# J2V8 (attached code / avr8js_demo_vanilla/run-v8.js):
Program running...
Time: Blink finished in 0 ms
Time: Blink finished in 699 ms
Time: Blink finished in 613 ms
Time: Blink finished in 646 ms
Time: Blink finished in 620 ms
Time: Blink finished in 611 ms
Time: Blink finished in 604 ms
Time: Blink finished in 604 ms
Time: Blink finished in 631 ms
Time: Blink finished in 622 ms
Time: Blink finished in 604 ms

# NodeJS: v13.12.0
> node run.js

Time: Blink finished in 1 ms
Time: Blink finished in 162 ms
Time: Blink finished in 113 ms
Time: Blink finished in 114 ms
Time: Blink finished in 118 ms
Time: Blink finished in 112 ms
Time: Blink finished in 111 ms
Time: Blink finished in 132 ms
Time: Blink finished in 110 ms
Time: Blink finished in 114 ms
Time: Blink finished in 116 ms
Time: Blink finished in 112 ms
Time: Blink finished in 112 ms

# graalvm
> /opt/graalvm-ce-java8-20.1.0/bin/node run.js 

Program running...
Time: Blink finished in 1 ms
Time: Blink finished in 2353 ms
Time: Blink finished in 2190 ms
Time: Blink finished in 783 ms
Time: Blink finished in 767 ms
Time: Blink finished in 802 ms
Time: Blink finished in 848 ms
Time: Blink finished in 803 ms
Time: Blink finished in 759 ms
Time: Blink finished in 825 ms
Time: Blink finished in 755 ms
Time: Blink finished in 804 ms
Time: Blink finished in 788 ms
Time: Blink finished in 778 ms

---

```java
package javascript.j2v8;

import java.io.File;
import java.io.IOException;
import java.nio.file.Files;
import java.nio.file.Path;
import java.nio.file.Paths;

import com.eclipsesource.v8.NodeJS;
import com.eclipsesource.v8.V8;

public class J2V8 {
  public static void main(String[] args) throws IOException {
    
    Path path = Paths.get("/media/ricardo/Dados/TEMP/testes-java/avr8js_demo_vanilla/run-v8.js");

    NodeJS node = NodeJS.createNodeJS();
    node.exec(path.toFile());

    while (node.isRunning()) {
      node.handleMessage();
    }
  }
}
```

ricardojlrufino, 2020-06-07:
# Running on: browser
bundle.js:2522 Fetch finished loading: GET "http://localhost:3000/blink.hex".
bundle.js:2560 Program running...
bundle.js:2605 Time: Blink finished in 0 ms
bundle.js:2605 Time: Blink finished in 244 ms
bundle.js:2605 Time: Blink finished in 183 ms
bundle.js:2605 Time: Blink finished in 224 ms
bundle.js:2605 Time: Blink finished in 170 ms
bundle.js:2605 Time: Blink finished in 168 ms
bundle.js:2605 Time: Blink finished in 166 ms
bundle.js:2605 Time: Blink finished in 172 ms
bundle.js:2605 Time: Blink finished in 167 ms
bundle.js:2605 Time: Blink finished in 192 ms
bundle.js:2605 Time: Blink finished in 164 ms
bundle.js:2605 Time: Blink finished in 170 ms
bundle.js:2605 Time: Blink finished in 167 ms
bundle.js:2605 Time: Blink finished in 166 ms
bundle.js:2605 Time: Blink finished in 177 ms
bundle.js:2605 Time: Blink finished in 166 ms
bundle.js:2605 Time: Blink finished in 165 ms
bundle.js:2605 Time: Blink finished in 169 ms
bundle.js:2605 Time: Blink finished in 164 ms
bundle.js:2605 Time: Blink finished in 165 ms
bundle.js:2605 Time: Blink finished in 179 ms
bundle.js:2605 Time: Blink finished in 175 ms
bundle.js:2605 Time: Blink finished in 164 ms
bundle.js:2605 Time: Blink finished in 178 ms

urish, 2020-06-08:
What if you remove the "TaskScheduler" and make `AVRRunner.execute()` run the loop a larger number of times? (let's say 16,000,000, so you get exactly one second of simulation?)

ricardojlrufino, 2020-06-08:
Could you do this ?
The branch: https://github.com/ricardojlrufino/avr8js_demo_node/tree/vanilla

ricardojlrufino, 2020-06-08:
For comparison purposes I decided to share with you the results I obtained using
Native JAVA -> C integration using SIMAVR
[JAVA] Time  change: 38ms                                                                                                                                                                                           
[JAVA] Time  change: 37ms                                                                                                                                                                                       
[JAVA] Time  change: 38ms                                                                                                                                                                                         
[JAVA] Time  change: 38ms

urish, 2020-06-08:
Interesting! 

I have sent you a PR which disables timer1/2, I wonder how it compares with your previous measurements

ricardojlrufino, 2020-06-09:
Improved Now
```
Running on: NodeJS
Program running...
Time: Blink finished in 1 ms
Time: Blink finished in 104 ms
Time: Blink finished in 67 ms
Time: Blink finished in 76 ms
Time: Blink finished in 66 ms
Time: Blink finished in 71 ms
Time: Blink finished in 67 ms
Time: Blink finished in 65 ms
Time: Blink finished in 66 ms
Time: Blink finished in 73 ms
Time: Blink finished in 66 ms
```

But dont respect to delay() like java version..
was the timer doing this function?
```C++
      void loop() {
        digitalWrite(LED_BUILTIN, HIGH);
        Serial.println("V1");
        delay(100);
        digitalWrite(LED_BUILTIN, LOW);
        Serial.println("V0");
        delay(100);
      }
```

urish, 2020-06-09:
Timer 0 is used for delay. Timer 1/2 shouldn't make any difference with regards to `delay()`.

We may be able to improve timer speed once #38 (and other related improvements) will be implemented

ricardojlrufino, 2020-06-09:
I achieved the promising result using SIMAVR, which is easier to integrate without suffering from performance problem, and more portable.
I'm developing a plugin for the Arduino IDE
Demonstration of integration (BR audio, sorry...)
https://youtu.be/58tEhkvaeOk

my Arduino IDE
https://github.com/ricardojlrufino/Arduino

urish, 2020-06-10:
Very nice work @ricardojlrufino :-) I see that you also added auto complete and other useful features to the Arduino IDE, that's awesome! What are your next plans?

Yes, it seems that SIMAVR is a better tool for the job. AVR8js was built with the web browser/Node.js environment in mind. There's an ongoing experiment for using Web Assembly (#35), but it might take some time.

ricardojlrufino, 2020-06-10:
> What are your next plans?
Create a more robust development environment (with file explorer) (https://github.com/ricardojlrufino/JExplorerTree) and improved layout (see in video: https://youtu.be/58tEhkvaeOk)
And this HMI builder:
[image]

Then I will re-implement the Autocomplete, as it was made for version 1.5.x, the version that was integrated into to official beta version (and stopped),  didn’t stay with usability and poor performance, because includes other tools .

I'm looking for collaborators, if only to test the improvements

urish, 2020-06-20:
That looks awesome Ricardo!

ricardojlrufino, 2020-06-21:
Alfa release:
https://github.com/ricardojlrufino/RITDevX

Docs:
https://opendevice.atlassian.net/wiki/spaces/RITDevX

closed 2020-07-29 by urish