wokwi/avr8js #51: usart.ts code typo?

yiivon, 2020-07-09:
src/peripherals/usart.ts line 103
` if (ucsrb & UCSRA_TXC && ucsrb & UCSRB_TXCIE) {`
is `ucsra & UCSRA_TXC` ?

urish, 2020-07-09:
So it seems! 

How did you come across this issue?

yiivon, 2020-07-09:
> So it seems!
> 
> How did you come across this issue?

I am trying to implement the Rx part hardly and ..., so, you see

urish, 2020-07-09:
The Rx part for AVR8js? as in issue #11 ?

yiivon, 2020-07-09:
> The Rx part for AVR8js? as in issue #11 ?

Yes

urish, 2020-07-09:
That's great!

Would you be interested in sending a PR to fix this issue as well?

In addition to fixing the actual issue, a test case which would fail before the fix will be very helpful

yiivon, 2020-07-09:
I am not very familiar with avr usart, fortunately I seem to have simply implemented the Rx part of it. However, I am not very sure whether this implementation will bring about other undesirable problems, hope helping and  thank you.
[image]
[image]

urish, 2020-07-09:
Got it, thanks for reporting!

What are you building?

commit 6ee7d2f2 (2020-07-09): fix(usart): TXC interrupt triggered incorrectly

close #51
closed 2020-07-09 by urish