wokwi/avr8js #55: Avr8js does not to run grbl firmware completely accurately.

yiivon, 2020-07-24:
In fact, I am very eager for avr8js to run [grbl](https://github.com/gnea/grbl) [firmware](https://github.com/gnea/grbl/releases/download/v1.1h.20190825/grbl_v1.1h.20190825.hex) completely and accurately.

GRBL is a firmware implemented in pure C language running on arduino uno (ATmega328P).

I use the latest version of Proteus to run grbl for arduino simulation, and I can get the correct result. But the current avr8js does not seem to be able to achieve the same, I guess it may be that some implementations of timer are not perfect.

If you also want to make avr8js run like real chips, you can try to run grbl firmware in emulation, which may solve this problem and make avr8js close to real chips again. At the same time, it also helped me a lot.

Ask for your help and thank you sincerely!

_Originally posted by @yiivon in https://github.com/wokwi/avr8js/issues/11#issuecomment-663224127_

urish, 2020-07-24:
Thanks @yiivon! 

Can you give more details about your setup?

What simulated hardware did you use? How did you test if you get correct result from the GRBL firmware?

urish, 2020-08-01:
I've just released a fix for issue #41. If the original issue was related to timers, the fix may help.

yiivon, 2020-08-05:
> I've just released a fix for issue #41. If the original issue was related to timers, the fix may help.

Thank you for your great work, it is very useful for everyone who loves avr programming. 

I have been working on solving some mechanical problems these days, so I didn’t reply to your message in time. 

I’m so sorry and so grateful.

urish, 2020-09-02:
Hi @yiivon, I'm closing the issue for now, but if this is still a problem, feel free to add more details (about how you tested for correct results) and reopen.

closed 2020-09-02 by urish

drf5n, 2023-03-31:
I think it is still a problem-- See https://github.com/wokwi/avr8js/issues/122