wokwi/avr8js #60: Timing issues

arcostasi, 2020-09-10:
Hi @urish I'm running the code below, and I noticed that the millis() function goes crazy after running for a while, approximately 1 ~ 2 min, it happened in my tests with Electron and in the online simulation.

Have any idea what it could be? Thanks

tested in the environment: https://wokwi.com/arduino/libraries/demo/blink

diagram.json
```
{
  "version": 1,
  "author": "Anderson Costa",
  "editor": "wokwi",
  "parts": [
    { "id": "uno", "type": "wokwi-arduino-uno", "top": 120, "left": 20 },
    { "id": "r1", "type": "wokwi-resistor", "top": 67, "left": 80, "rotate": 90, "attrs": {"value": "220"} },
    { "id": "l1", "type": "wokwi-led", "left": 80, "top": 0, "attrs": { "color": "red" }},
    { "id": "r2", "type": "wokwi-resistor", "top": 67, "left": 115, "rotate": 90, "attrs": {"value": "220"} },
    { "id": "l2", "type": "wokwi-led", "left": 120, "top": 0, "attrs": { "color": "green" }},
    { "id": "r3", "type": "wokwi-resistor", "top": 67, "left": 163, "rotate": 90, "attrs": {"value": "220"} },
    { "id": "l3", "type": "wokwi-led", "left": 160, "top": 0, "attrs": { "color": "blue" }}
  ],
  "connections": [
    ["uno:GND.1", "l1:C", "black", []],
    ["uno:GND.1", "l2:C", "black", []],
    ["uno:GND.1", "l3:C", "black", []],
    ["r1:1", "l1:A", "red", []],
    ["r2:1", "l2:A", "green", []],
    ["r3:1", "l3:A", "blue", []],
    ["uno:13", "r1:2", "red", []],
    ["uno:11", "r2:2", "green", []],
    ["uno:8", "r3:2", "blue", []]
  ]
}
```

blink.ino
```
#define DELAY_LED 1000

// LEDs connected to pins...
byte ledPins[] = {13, 11, 8};
byte i = 0;

unsigned int timerLed = 0;

bool switchLed = true;

// The setup function runs once when you press reset or power the board
void setup() {
  // Initialize digital pins LED as an output.
  for (byte n = 0; n < sizeof(ledPins); n++) {
    pinMode(ledPins[n], OUTPUT);
  }
}

// The loop function runs over and over again forever
void loop() {
  if ((millis() - timerLed) > DELAY_LED) {
    timerLed = millis();

    digitalWrite(ledPins[i], switchLed);

    if ((i + 1) == sizeof(ledPins)) {
      switchLed = !switchLed;
    }

    i = (i + 1) % sizeof(ledPins);
  }
}
```

urish, 2020-09-10:
Thanks for reporting!

Actually, it seems like the issue is in the code itself: the type of `timerLed` is unsigned int, which goes from 0 to 65535. 

Can you try changing the type of this variable to `unsigned long` or `uint32_t` and see if this fixes the issue?

p.s. thanks for posting these super-clear reproduction instructions along with the diagram file!

arcostasi, 2020-09-10:
Worked perfectly @urish 👍 

I'm sorry, for a moment I thought it could be in the simulation, I'm happy to know that it was an error in my code! :) Thanks

urish, 2020-09-10:
You're welcome :-)

closed 2020-09-10 by urish