wokwi/avr8js #63: IOT project with wokwi

shashankkumar88, 2020-10-24:
Hi @urish , I am building a IOT project which requires wifi connectivity. Does any of the Arduino board in wokwi has wifi built in? Or what would be the way to get wifi connectivity in wokwi simulator?

urish, 2020-10-24:
Hello! What project are you building?

Currently, no WiFi support is available. This is something I may look into in the future!

closed 2020-10-24 by urish

shashankkumar88, 2020-10-24:
> Hello! What project are you building?
> 
> Currently, no WiFi support is available. This is something I may look into in the future!

I am sending temperature sensor data to ThingSpeakIOT for analytics

shashankkumar88, 2020-10-26:
Hello! How can we get the React Source code for https://wokwi.com/playground/lcd1602-i2c?utm_source=github&utm_medium=social&utm_campaign=wokwi-playgrounds
I am planning to retrieve data from ThingSpeak and display on LCD. As the wifi is not present in  UNO board, I am planning to perform the data retrieval in the React code and pass it on to Arduino simulator dynamically.

urish, 2020-10-27:
Hello! You can find a copy of the LCD1602 controller here: https://github.com/arcostasi/avr8js-electron-playground/blob/master/src/shared/lcd1602.ts

Alternatively, you can also try to use a custom iframe inside wokwi.com. It's not documented yet, but here is an example which uses Three.js to display a rotating smart LED cube: https://wokwi.com/arduino/projects/278306947843752461

If you are interested in this direction, I can also prepare an example with the LCD screen inside a custom iframe

shashankkumar88, 2020-10-27:
But for this how do I get the React project?I mean the code sandbox similar to this https://codesandbox.io/s/arduino-simulator-js-9z159?file=/src/App.js. Also I was following your tutorial here https://www.youtube.com/watch?v=fArqj-USmjA but the docs panel in the wokwi elements catalogue is empty for all the elements. For example https://elements.wokwi.com/?path=/docs/led--red

urish, 2020-10-27:
Yes, the docs has been broken for sometime, I haven't been able to figure out why. I tried upgrading Storybook to v6, which fixes the docs tab, but it introduced other issues. Anyway, you can have a look at the docs from that work-in-progress branch:

https://deploy-preview-41--wokwi-elements.netlify.app/?path=/docs/lcd1602--hello-world

If you want I can try to come up with a complete example on codesandbox, but it will probably take me a few days to get to it...

Fi1osof, 2025-09-04:
> Hello! What project are you building?
> 
> Currently, no WiFi support is available. This is something I may look into in the future!

Future is here :-) Is WiFi support is available?