wokwi/avr8js #66: Mega example only renders first frame on 0.12

elliottkember, 2020-11-15:
I'm trying to update to the latest code, but just can't figure out how to make it work right :)

Going to keep investigating and see if I can figure out what changed.

https://stackblitz.com/edit/avr8js-mega-ws2812-mdzn4x?file=package.json

urish, 2020-11-15:
From my testing so far, it seems like it gets broken starting from version 0.11.1 (0.11.0 works fine), can you confirm if you are observing the same?

urish, 2020-11-15:
I think I spotted it - can you try changing the value of `FLASH` in `execute.ts` to `0x40000` and see if it helps?

elliottkember, 2020-11-16:
> I think I spotted it - can you try changing the value of FLASH in execute.ts to 0x40000 and see if it helps?

That works! It seems to be running quite slowly. I'll debug that and see whether that's something in my code.

urish, 2020-11-16:
Lovely!

closed 2020-11-16 by urish