wokwi/avr8js #73: Not clear what arduino boards and what microchips exactly can be simulated with this library ?

divinciest, 2020-12-12:
Hello.
First thanks for the effort.

The company that I work for is initiating a project that heavily depends on simulating boards , and Arduino is in our list. as in this project https://github.com/buserror/simavr , it is clear what it is capable of simulating in exact technical terms. a list of microchips.
But for this project it seemed to me rather vague in terms of what technical specification to expect from the serial I/O and clock speed and instruction set stand point...

I would appreciate an explanation of :

1. What list of boards can this library simulate ?  
2. What microchips/instruction set does it expect ?
3. What pins are supported and it is possible to add more ?
4. To what extinct this library is customizable to add support for custom technical specifications ?

Thanks

urish, 2020-12-12:
Hi @divinciest, I see you are working on a robot operating system called Merat. That sounds very interesting, if you have a few moments to describe the project, I'd love to learn what you are working on.

Regarding AVR8js: the library simulates AVR 8-based chips, currently focusing on the ATmega328p. ATtiny85/45/25 and ATmega2560 are partially supported (but the support is already good enough to run most Arduino programs). You can find some examples in the [README](https://github.com/wokwi/avr8js/edit/master/README.md).

All the peripherals of the chips are fully configurable, so adding new chips shouldn't be too hard. It is usually a matter of defining the flash / EEPROM / SRAM size, the memory addresses of the peripherals, and the interrupt vectors.

In terms of boards, it does not simulate directly any board, only the chips, so you are free to define the pin assignments in any way you want. For instance, the Arduino UNO uses at ATmega328p, and at Arduino MEGA uses the ATmega2560. You can find both boards on [wokwi.com](https://wokwi.com/arduino/libraries/demo), and it should be pretty straightforward to add new boards as long as they use a supported AVR8 chip.

Is your simulation expected to run in a web browser? Or are you using electron / other JavaScript based technology to build your app? I don't have too much experience with simavr, but the aim of AVR8js is to bring the AVR simulation to the world JavaScript, so when choosing a library I advise to take into account what environment you'll be running on. If JavaScript/TypeScript based, then AVR8js could be a good fit. If native C/C++, then simavr would probably be a better choice.

divinciest, 2020-12-15:
We are on the research and documentation stage, and we are still finishing a preceding project, I  will respond as soon as this becomes our full time project which is soon.

urish, 2020-12-20:
Meanwhile, I [updated the README](https://github.com/wokwi/avr8js#which-chips-can-be-simulated) to explain more about the supported chips

septatrix, 2022-01-01:
I think it would also be feasible to compile simavr to webassembly in which case this library could simply become a wrapper around simavr.
This would directly come with support for many different chips, peripherals (e.g. I2C) which are unsupported according to the wiki) and a decent performance boost.

See discussion in #49 for how big the performance boost might be. Also #35 tasks about AssemblyScript, WebAssembly (different from the former) and maybe even AVR -> wasm transpiling to improve performance through they would require a larger effort which would not be required when writing a wrapper around simavr.

urish, 2022-01-01:
You are invited to give it a go and share the results!