wokwi/avr8js #88: how to get rid of slash

pfalstad, 2021-03-09:
This is probably something really simple, but why is there a leading slash on the src URLs generated when parcel build runs on the demo project?  After it runs, in demo/build/index.html, the reference to src.whatever.js or src.whatever.css all have leading slashes, which makes it impossible for the resulting output to be used inside of a larger tree.  I can remove them by hand or with a script each time I build, but I was wondering if there was a way to get rid of them.  Thanks.

urish, 2021-03-09:
I will try to see if there's some flag that tells parcel not to do it. 

In general, I hope to let go of parcel soon. Once I figure out a proper solution for preactjs/wmr#394, that'd probably be a goodbye for parcel...

pfalstad, 2021-03-09:
Oh never mind I found it.  Now that i knew it was parcel doing it (I thought it was the typescript compiler), I got lucky with a google search.

parcel build src/index.html --public-url ./

closed 2021-03-09 by pfalstad

urish, 2021-03-09:
Glad I was able to hint at the right direction :-)

Are you working on something like this?

https://markmegarry.github.io/AVR8js-Falstad/

pfalstad, 2021-03-09:
Yes I am working on that, fixing a few things and getting it to work without forking a new version of the simulator.

urish, 2021-03-09:
Lovely! If you need anything, I'm here (also on [discord](https://wokwi.com/discord), #avr-internals)

urish, 2021-03-11:
Thank you Paul :-) I'm happy to see this!