wokwi/avr8js #93: analogWrite() doesn't work?

pfalstad, 2021-05-31:
I went to your first two-led example on https://blog.wokwi.com/avr8js-simulate-arduino-in-javascript/ and changed it to the following:

void setup() {
  pinMode(LED_BUILTIN, OUTPUT);
}

void loop() {
  analogWrite(LED_BUILTIN, 128);
}

But it doesn't work; the green LED is consistently on.  If I change the 128 to 127, the LED is consistently off.  Did I mess something up?

urish, 2021-05-31:
Hi Paul, pin number 13 (which the LED_BUILTIN is connected to), does not support PWM. The only Arduino Uno pins that support PWM (and thus `analogWrite()`) are 3, 5, 6, 9, 10 and 11.

Also, the specific example in the blog post uses an archaic version of AVR8js, so I'm not even sure if PWM would work there.

There's an LED PWM example, linked from the README of this repo, which shows a how to use PWM in AVR8js. It adds LED to pin 11, as well as some logic on the JavaScript side to convert the PWM signal into LED brightness:

https://stackblitz.com/edit/avr8js-pwm?file=index.ts

You can also find a walkthrough of how the PWM code example was built [in this video](https://www.youtube.com/watch?v=7KYzPJ-tQno).

I hope this is helpful for you!

pfalstad, 2021-05-31:
Great!  Thanks!

closed 2021-05-31 by pfalstad