wokwi/avr8js #94: PWM disturbances when running code

JonasForssell, 2021-06-05:
Dear Sir,

Thank you for a fantastic effort!
I am using your Avr8js in combination with the Falstad simulatior to try out PWM control.

This code successfully initiates mode15 and generates an approx 30kHz PWM on pin 9
However any code I run i main() disturbs the PWM.
Is this correct? I am taking a chance here because I do not have hardware at hand but according to the datasheet, timer 1 should have quite high priority.

many thanks!
/Jonas

#include <avr/io.h>
#include <avr/interrupt.h>
#include <util/delay.h>
 
void initialize() {​​​​​​​​​​
  // add external voltage (Javascript) with name "pin n" to access output pins
  pinMode(9, OUTPUT);
  pinMode(6, INPUT);
 
  // Clear interrupts
  cli();
  // Reset TCCR1B from Arduino meddeling
  TCCR1B = 0;
  // Set fast PWM Mode with Top value as a frequency controller
  TCCR1A |= (1 << WGM10) | (1 << WGM11);
  TCCR1B |= (1 << WGM12) | (1 << WGM13);
  // Set toggle on compare match
  TCCR1A |= (1 << COM1A0);
  // Set TOP value which will reflect period length between 0 to 65536
  OCR1A = 120;
  // Set prescaler to 1 and start PWM
  TCCR1B |= (1 << CS10);
 
}​​​​​​​​​​
 
int main() {​​​​​​​​​​​​​​​​​
 
initialize();
 
while (1) {​​​​​​​​​​​​​​​​​
  int v_out;
  
v_out = digitalRead(6);
delay (1);
 
}​​​​​​​​​​​​​​​​​
 
}

urish, 2021-06-05:
Hi Jonas,

Are you using the latest version of AVR8js? How do you verify the PWM frequency?

In general, if you want to use `delay()` in your code, you should enable interrupts `cli()`, and also, shouldn't override `main()`, rather put your code inside `setup()` and `loop()`. The `delay()` function is part of the Arduino core and requires Timer 0 to be initialized in a specific way. When you define your own `main()`, the Arduino initialization code does not run and thus Timer 0 is not initialized and `delay()` won't work correctly.

JonasForssell, 2021-06-05:
Hej Urish,
Thanks for the quick reply.
I tried the exact same program on an Arduino Uno now without issues. The signal was rock solid.
I used the Arduino editor to compile and upload the sketch.

I will try an adjusted code in falstad AVR8js but still it could be worthwile investigate in any case.

Thanks
/Jonas

urish, 2021-06-05:
Hi Jonas, 

I'm trying to isolate the problem and created a small project that should help reproducing the issue with just AVR8js:

https://stackblitz.com/edit/avr8js-pwm-scope?file=index.ts

I'm observing the frequency value jumping back-and-forth between 39801Hz and 19900Hz, is that also the case when you are running this in the falstad environment?

urish, 2021-06-05:
Also, my bad, enabling interrupts is done with `sei()`, not `cli()`

JonasForssell, 2021-06-05:
Yes that corresponds quite well with what I see in falstad.
sometimes a pulse or gap is twice the length.

urish, 2021-06-05:
> This code successfully initiates mode15 and generates an approx 30kHz PWM on pin 9
> However any code I run i main() disturbs the PWM signal out from AVR8JS making it irregular.

Do you also have a variation of the code that generates the PWM wave correctly in the emulator? Or did I understand you incorrectly?

JonasForssell, 2021-06-05:
I have not managed to generate a stable pulse other than when the while loop is completely empty.

urish, 2021-06-05:
Ok, looking at the physical Uno output v.s. the emulated one, I can definitely see there's a problem.

The real one shows a steady PWM wave (captured with Saleae Logic 8):

[image]

While the emulated one seems to sometimes miss a transition and stay high/low for two periods (screen shot from PulseView):

[image]

and zooming-in on a good period:

[image]

v.s. a bad one:

[image]

So this is definitely a bug. I'm hoping to dig into it sometime next week!

urish, 2021-06-05:
For reference: I uploaded my simulator logic analyzer setup here: https://wokwi.com/arduino/projects/300601028188832265

You can view the logic analyzer output by starting the simulation, stopping it after a second, and then it will automatically download a "signal.vcd" file to your computer. Import this file into [pulseview](https://sigrok.org/wiki/PulseView) to view the signal.

JonasForssell, 2021-06-17:
Apologies for asking but have you made any progress urish?

urish, 2021-06-17:
Unfortunately not yet. But I plan to do so this weekend.

By the way, what project are you working on?

JonasForssell, 2021-06-18:
I am trying to model a resonant DC/DC converter in Falstad/avr8js. I want to try to make it digitally controlled by varying the frequency. 
The combination of your software and Falstad is very powerful and allows me to try out the concept before building anything in hardware. I am highly impressed by your software

urish, 2021-06-19:
That sounds really interesting! I'd love to see this in action :)

Regarding the issue - I've written a test case that reproduces it. Expect a solution soon!

```cpp
  it('should set the OCF1A flag when OCR1A == 120 and the timer overflowed past 120 in WGM mode 15 (issue #94)', () => {
    const cpu = new CPU(new Uint16Array(0x1000));
    new AVRTimer(cpu, timer1Config);
    cpu.writeData(TCNT1, 118);
    cpu.writeData(OCR1A, 120);
    cpu.writeData(OCR1B, 4); // To avoid getting the OCF1B flag set
    cpu.writeData(TCCR1A, WGM10 | WGM11); // WGM: Fast PWM
    cpu.writeData(TCCR1B, WGM12 | WGM13 | CS10); // Set prescaler to 1
    cpu.cycles = 1;
    cpu.tick();
    cpu.cycles = 5;
    cpu.tick();
    expect(cpu.readData(TCNT1)).toEqual(1);
    expect(cpu.data[TIFR1] & (OCF1A | OCF1B)).toEqual(OCF1A);
    expect(cpu.pc).toEqual(0);
    expect(cpu.cycles).toEqual(5);
  });
```

closed 2021-06-19 by urish

urish, 2021-06-19:
Fix released as avr8js 0.15.2. Now just need to wait for falstad to update :)

Meanwhile, I confirmed the fix on wokwi.com, seems to do the trick :)

commit a774df98 (2021-06-19): fix(timer): Timer1 PWM issues #94

close #94