wokwi/avr8js #98: else/if statements not working.

TheVoid3125, 2021-07-12:
Hello,
I am trying to create a "snackbox" with an Arduino Uno R3. I just started using this simulator (which, by the way, is amazing!). It worked very well, but when I tried to implement an else/if statement and run the program, it told me there were errors in the code! I am new to C++, so I looked up C++ else/if statements and it said they were possible, and I also looked up Arduino else/if statements, which also existed. By the way, if anyone reading this doesn't know what an else/if statement is, it looks like this:

if (**_condition 1_**) {
   **_statement1_**
} else if (**_condition 2_**) {
   **_statement2_**
} **_end, else statement, or else/if statement (can continue on and on)_**

I don't know how this is not possible in Wokwi, as it does use C++. For the time being, I will nest and if statement in the else statement, but I would like the option of an else/if statement in Wokwi at some point. I also would like to add that there is no photoresistor, which I am adding to my final build. I am substituting the photoresistor for a slide potentiometer in the simulator. Any fixes on these would be very appreciated!

[image]

TheVoid3125, 2021-07-12:
Nevermind! I just realized that I misspelled "position." if/else statements do work! I still would like a photoresistor though.

closed 2021-07-12 by TheVoid3125

urish, 2021-07-13:
Thanks for the feedback! Didn't Wokwi tell you that the error was in `postion` when you tried to run the program?

As for the photoresistor - would this kind do the trick?

[image]

TheVoid3125, 2021-07-13:
Wokwi did tell me the position, but I kinda skimmed over it and didn't actually take in what it was telling me... I didn't mention it before because I was too embarrassed. 😳 As for the photoresistor, yes that would do the trick, as it is a photoresistor, but there are more simple ones:

[image]

The photo you sent me is a special type of photoresistor that you can change the photoresistance manually, which can be easier than programming the photoresistance. Also, I would like to "join the club" with that subscription, but seeing as I have no income currently, I can't really guarantee that I could keep paying.

urish, 2021-07-27:
The LDR photoresistor module is now ready: https://docs.wokwi.com/parts/wokwi-photoresistor-sensor

Thanks for considering the club! I'm sorry to hear that you have no income currently, I truly hope the situation will get better!