wokwi/avr8js #99: Not Saving Correctly

TheVoid3125, 2021-07-13:
Hello again, this is a different issue than before:
Wokwi sometimes doesn't save correctly, causing me to lose a lot of hard work. I have been restarting my computer to download a **TRUSTED** antivirus in response to fake virus warnings (any help on these warnings would also help, they are driving me crazy!), but I don't know how restarting the computer could erase some of my progress...

urish, 2021-07-13:
Don't you see the project when you go to https://wokwi.com/arduino/projects ?

TheVoid3125, 2021-07-13:
Yes, I still have most of the project, but it doesn't go to the save I make before restarting my computer.

urish, 2021-07-13:
Is this something that happens frequently? 

If I find a way to reproduce it on my end, I'll be able to look for a solution

TheVoid3125, 2021-07-13:
Maybe it is the fact that I saved, and then restarted my computer?

TheVoid3125, 2021-07-13:
It seems to open the save I made previous to the most recent save.

urish, 2021-07-13:
It could be - saves may take a few seconds to arrive at the server, so if the browser tab closed before the save request hit the server, it makes that it didn't make it

TheVoid3125, 2021-07-13:
Maybe? But I transferred to a new browser to avoid the virus warnings, saved on the old browser, than immediately opened the project in the new browser. But maybe that was because the project was saved on my computer?

urish, 2021-07-13:
The projects are not saved on the local computer. If you attach a link to the project in question, I can try to take a look at the server logs and see if there are any other versions in the server logs

TheVoid3125, 2021-07-13:
Ok

https://wokwi.com/arduino/projects/303949867923800642

closed 2021-07-13 by TheVoid3125