wokwi/avr8js #102: Timer outputs are not writen back into PIN Register

fjangfaragesh, 2021-08-13:
Example:

The status of each pin should always be transferred to the PIN "register".  
Example:
Timer1 sets `PB1` to `HIGH`, `PINB` should be `0bxxxxxx1x`,  
Timer1 sets `PB1` to `LOW`, `PINB` should be `0bxxxxxx0x`,  

my test code:
```cpp
int main(void){
  DDRB |=  (1 << PORTB1);
  TCCR1A |= (1 << COM1A1) | (1 << WGM10);
  TCCR1B |= (1 << CS12) | (1 << CS10) | (1 << WGM12);
  Serial.begin(9600);
  while(1) {
    OCR1A = 255;// PB1 on
    for (int i = 0; i < 10; i++) {
      _delay_ms(100);
     Serial.println(PINB);
     Serial.flush();
    }
    OCR1A = 1;// PB1 off (most of the time)
    for (int i = 0; i < 10; i++) {
      _delay_ms(100);
      Serial.println(PINB);
      Serial.flush();
    }
  }
}
```
Using a real Arduino, the serial output is: 2 2 2 2 2 ... 0 0 0 0 0 ... 2 2 2 ... ...  
In simulation the serial output is: 0 0 0 0 0 0 0 ...

(tested with Arduino Nano in real and Arduino Uno in simulation)

urish, 2021-08-13:
That indeed sounds like a bug. Thanks for reporting, I'll investigate!

closed 2021-08-13 by urish

urish, 2021-08-13:
Found it! 

Should be fixed now :)

commit d98b8fe7 (2021-08-13): fix(gpio): timer outputs not reflected in PIN register #102

fix #102