wokwi/avr8js #103: SBI / CBI work differently from hardware

urish, 2021-09-03:
SBI / CBI sometimes produce incorrect result when used with special I/O registers, such as the PIN registers.

The following Arduino program demonstrates the issue:

```cpp
void setup() {
  DDRD = bit(3) | bit(6);
  PIND |= bit(3) | bit(6);
}

void loop() {
  delay(1000);
  PIND |= bit(6);
}
```

When running [in simulation](https://wokwi.com/arduino/projects/308744248659804737), the second line of `loop()` will drive pin 3 LOW *in addition* to toggling pin 6. This is in contrast with the physical hardware, where pin 3 (the red LED) in unaffected:

https://user-images.githubusercontent.com/892318/132061911-7b258248-aa2b-4297-900f-ee23c0e8f8c9.mp4

By looking at the assembly listing, you can see that `PIND |= bit(6)` compiles to: 

```
sbi	0x09, 6	; 9
```

rcwilcox, 2021-09-04:
Just to exonerate the PIN Register simulation, I verified that writing PINx without using SBI works fine in Sim.

e.g.
  
   ldi r18,0b00000100    ;  Set B2      
   out PIND,r18              ;  Toggle PD2

closed 2021-09-06 by urish

urish, 2021-09-09:
And now, a video version of how I fixed this bug: 
https://youtu.be/r_ZIiDi8nCU

commit 5bc85ec0 (2021-09-06): fix(gpio): CBI/SBI handling in writes to PIN register #103

fix #103