wokwi/avr8js #112: How does wokwi playground validate the circuit?

Sharvin26, 2021-11-09:
Hello @urish,

I have been following this project for some time and was wondering how does the **Wokwi Playground** validate the circuit?

For reference I have created an example on codesandbox with two elements:

1. Wokwi led
2. Wokwi Arduno Mega

The port C of wokwi led is connected to 7th pin of  Wokwi Arduno Mega and when I click on run button and it runs clicks which is wrong.

https://codesandbox.io/s/floral-cherry-h23d1?file=/src/App.js:259-487

In **Wokwi Playground** when I create scenario the same scenario it doesn't work which is correct.

https://wokwi.com/arduino/projects/314782342842417729

I want to know if there's any utility function that's making this possible? Does the avr8js library provide such utility function or it's inbuilt in the wokwi playground.

Any help is much appreciated. Thank you

urish, 2021-11-09:
Hi @Sharvin26 ! 

This behavior is indeed part of Wokwi. The LED simulation looks at both connections of the LED, and the LED goes on only if the negative pin (cathode) is digital 0 and the positive pin (anode) is digital 1. It also checks the both pins are driven (i.e. connected to a voltage pin or a GPIO pin in output mode), applies [gamma correction](https://docs.wokwi.com/parts/wokwi-led#gamma-correction), and averages the signal to support PWM output.

Here's a [sample implementation of LED PWM support](https://www.youtube.com/watch?v=7KYzPJ-tQno).

closed 2021-11-09 by urish

Sharvin26, 2021-11-10:
Hi @urish,

Thanks for the response. 

So basically it is not available under avr8js library and needs to be implemented manually.

I saw the demo that you shared but I couldn't find anywhere you are checking if LED has negative and positive pin connected. Same for GPIO and voltage pin )

I think I might have missed something in the example/code.

Any help is much appreciated. Thank you.

urish, 2021-11-20:
Hi Sharvin, the example only covers PWM, not other parts of the simulation. Checking the connections depends on how you define the connections, and what kind of simulation you are doing.

For example, there's an example where [AVR8js is paired with Falstad's CircuitJS1](https://www.falstad.com/circuit/avr8js/), and they have full analog simulation there. So you can do things like resistor dividers and other analog stuff that Wokwi can't do. 

The sum up, AVR8js does not implement any kind of circuit simulation, it only simulates the processor. It's up to the user to decide if/how to simulate additional hardware and the circuit.