wokwi/avr8js #117: Simulator not masking 16 bit timer register writes in fixed-TOP modes?

drf5n, 2022-02-21:
Per https://forum.arduino.cc/t/why-doesnt-analogwrite-protect-its-writes-to-16bit-registers/961470/5?u=davex  I think I found a difference between the silicon and wokwi behavior in masking off the top bits of 16 bit register writes in fixed TOP modes.

Here's a sim: https://wokwi.com/arduino/projects/324214622608425554

Output on a Mega is:

```
TCCR0A 0b11 TCCR0B 0b11 OCR0A 0 OCR0B 0
TCCR1A 0b10000001 TCCR1B 0b11 TCCR1C 0b0 OCR1A 255 OCR1B 0
On real mega, OCR1A is 255
On Wokwi mega, OCR1A is 65535
```
on Wokwi is
```
TCCR0A 0b11 TCCR0B 0b11 OCR0A 0 OCR0B 0
TCCR1A 0b10000001 TCCR1B 0b11 TCCR1C 0b0 OCR1A 65535 OCR1B 0
On actual mega, OCR1A is 255
On Wokwi mega, OCR1A is 65535
```

The relevant section of the https://ww1.microchip.com/downloads/en/DeviceDoc/Atmel-7810-Automotive-Microcontrollers-ATmega328P_Datasheet.pdf datasheet is this bit from 15.9.4:

[image]

urish, 2022-02-21:
Thanks for the super detailed report!

Fixed + released a new version. Your test project now shows "255", just like the Mega would!

closed 2022-02-21 by urish

drf5n, 2022-02-21:
I see it works properly for timers 3,4,5 as well.  Thanks.

urish, 2022-02-21:
Great :)