wokwi/avr8js #120: HELP Using LCD

ricardojlrufino, 2022-03-25:
Hi @urish 
do you have an example of how to use the LCD with Avr8js?
https://elements.wokwi.com/?path=/story/lcd1602--hello-world

urish, 2022-03-25:
Hi @ricardojlrufino, I think there's an example inside [avr8js-electron-playground](https://github.com/arcostasi/avr8js-electron-playground)

ricardojlrufino, 2022-03-25:
I see this project but it can't run:
https://github.com/arcostasi/avr8js-electron-playground/issues/5

ricardojlrufino, 2022-03-25:
Where are "driver" part for https://github.com/wokwi/wokwi-elements ?

urish, 2022-03-25:
I hope that Anderson will reply soon to the issue, I believe he's quiet busy nowadays...

The "driver" part is not open, but some drivers are available in the [AVR8js examples](https://github.com/wokwi/avr8js#unofficial-examples), e.g. the [SSD1306](https://stackblitz.com/edit/avr8js-ssd1306).

I searched the avr8js electron playground, and here's the source code for the LCD1602 driver:

https://github.com/arcostasi/avr8js-electron-playground/blob/master/src/shared/lcd1602.ts

I'm working on a way for users to write custom drivers, so that might lead that way to opening additional drivers in the future. If you are interested, I just opened a new #custom-parts channel on the [Discord chat](https://wokwi.com/discord).

ricardojlrufino, 2022-03-25:
It's giving a compilation error, but I think I can solve it another way. Thanks.
 ```
/sketch/sketch.ino:3:10: fatal error: Adafruit_SSD1306.h: No such file or directory
#include <Adafruit_SSD1306.h>
```

I'm working on a tool to be able to integrate the Arduino IDE into the browser, so you can do the local compilation, and just the simulation using Avr8js, sending the HEX using websocket (no internet required)

The possibility to use only the browser (as an IDE) is fantastic, but there are some limitations...

urish, 2022-03-26:
As for the compilation error: You'd need to send a "libraries.txt" file to the build server, telling it to use the Adafruit_SSD1306 library, similar to how it works on Wokwi.com. In the past, the build server used to have all the libraries preinstalled, but nowadays the users have to specify which libraries they use.

The tool sounds very useful. We will probably have some solution for local compilation on Wokwi.com in the future as well - some users have requested this.

closed 2023-01-06 by urish