wokwi/avr8js #121: Compilation for #define BOUNCE_LOCK_OUT #include <Bounce2.h> odd?

drf5n, 2022-03-28:
Maybe there's a bug in the simulator's compilation order per  https://github.com/thomasfredericks/Bounce2/issues/85#issuecomment-1080661195   versus https://wokwi.com/projects/324855073217708626   

```
// Try Bounce2's lockout form
// (Doesn't work, 2022-03-26 -- Issue https://github.com/thomasfredericks/Bounce2/issues/85 )
#define BOUNCE_LOCK_OUT // per https://forum.arduino.cc/t/leading-edge-debouncing/964407/15?u=davex

#include <Bounce2.h> // https://github.com/thomasfredericks/Bounce2
Bounce b = Bounce(); // Instantiate a Bounce object
```

The code acts as if the #include <Bounce2.h> was read before the #define BOUNCE_LOCK_OUT statement was processed, and does not do the conditional processing as expected.

In the sim, the Bounce2 scheme should be catching the leading edge of noisy button presses at the same time as the leading edge and Wasser schemes.

urish, 2022-03-28:
Thanks for reporting! 

Can you try compiling the same sketch locally on your computer and then load the compiled .hex file into the simulation by pressing "F1" in the editor and choosing "Load HEX File and Start Simulation"?

drf5n, 2022-03-28:
Huh -- I didn't know about the F1 menu or that you could upload your own  compiled code.   But nevermind -- I had the same sort of issue with the externally compiled hex file, and simpler examples.  It seems like the library is loaded before the sketch and doesn't pay attention to sketch-defined defines:  https://github.com/thomasfredericks/Bounce2/issues/85#issuecomment-1081131688

I tried moving a copy of the library into the simulator's files, and a conditional `#error` planted in the .cpp file would pick up on the define in the Bounce2.h file but not from in the sketch.

The simulator acts like reality.

closed 2022-03-28 by drf5n

urish, 2022-03-28:
As someone on [discord](https://wokwi.com/discord) recently stated:

> That F1 function has all the good stuff !