wokwi/avr8js #125: Timer: Implement Input Capture

urish, 2022-05-02:
see wokwi/wokwi-features#352 for details

drf5n, 2023-04-19:
I see that input capture is mentioned on:

  https://docs.wokwi.com/parts/wokwi-arduino-mega

urish, 2023-04-19:
Yes :)

drf5n, 2024-06-26:
The missing Input Capture Interrupt infrastructure seems to cause the CTC modes that use ICRx as TOP to not be able to trigger the TIMERn_CAPT_vect at TOP.

Per https://ww1.microchip.com/downloads/en/DeviceDoc/Atmel-7810-Automotive-Microcontrollers-ATmega328P_Datasheet.pdf page 101:

>An interrupt can be generated at each time the counter value reaches the TOP value by either using the OCF1A or ICF1 flag according to the register used to define the TOP value. If the interrupt is enabled, the interrupt handler routine can be used for updating the TOP value. However, changing the TOP to a value close to BOTTOM when the counter is running with none or a low prescaler value must be done with care since the CTC mode does not have the double buffering feature. If the new value written to OCR1A or ICR1 is lower than the current value of TCNT1, the counter will miss the compare match. The counter will then have to count to its maximum value (0xFFFF) and wrap around starting at 0x0000 before the compare
match can occur. In many cases this feature is not desirable. An alternative will then be to use the fast PWM mode using OCR1A for defining TOP (WGM13:0 = 15) since the OCR1A then will be double buffered.

I think part of the missing bits might be around:

https://github.com/wokwi/avr8js/blob/6ecba64687e10113ad697860b5f103746a8ef329/src/peripherals/timer.ts#L315-L343