wokwi/avr8js #135: How to simulate DHT22 Temperature and humidity sensor

xucheng2008, 2023-01-06:
How to simulate DHT22 Temperature and humidity sensor?

urish, 2023-01-09:
Hello, are you asking about using the DHT22 sensor on Wokwi, or how to write a simulation model for this sensor for your own project built on top of avr8js?

xucheng2008, 2023-01-10:
I want to realize DHT22 sensor simulation on the basis of avr8js,but how to simulate DHT22 and send data to arduino uno? I don't know how to solve it!

urish, 2023-01-10:
To add DHT22, you'll have to implement the protocol. You can find the exact details in the datasheet: https://files.seeedstudio.com/wiki/Grove-Temperature_and_Humidity_Sensor_Pro/res/AM2302-EN.pdf

Specifically, the following diagram tells you how to encode the humidity and temperature values in a 40-bit signal:

[image]

xucheng2008, 2023-01-11:
Thanks for the information，But I have a question about the time the DHT sends the data accordingly. If the port listener detects host signals and sends DHT signals, the program is occupied for a long time. As a result, the main program cannot receive DHT signals。runner.portD.addListener Part of the code is as follows：
` if (firstFlag) {
      // runner.cpu.writeData(runner.portD.portConfig.DDR, 0);
      // runner.portD.setPin(2, true);
      runner.cpu.writeData(TCCR0B, CS01 | CS00);
      runner.cpu.writeData(TCNT0, 0);

      let currentTime = 0;
      lastTime = currentTime;
      runner.cpu.writeData(runner.portD.portConfig.DDR, 0);
      runner.portD.setPin(2, false);
      while (currentTime - lastTime < 80) {
        runner.cpu.tick();
        runner.cpu.cycles += 64;
        currentTime = runner.cpu.readData(TCNT0) * 4;
      }

      runner.cpu.writeData(runner.portD.portConfig.DDR, 0);
      runner.portD.setPin(2, true);

      runner.cpu.writeData(TCCR0B, CS01 | CS00);
      runner.cpu.writeData(TCNT0, 0);
      currentTime = 0;
      lastTime = currentTime;
      while (currentTime - lastTime < 80) {
        runner.cpu.tick();
        runner.cpu.cycles += 64;
        currentTime = runner.cpu.readData(TCNT0) * 4;
      }

      runner.cpu.writeData(runner.portD.portConfig.DDR, 0);
      runner.portD.setPin(2, false);

      for (let i = 0; i < 40; i++) {
        runner.cpu.writeData(TCCR0B, CS01 | CS00);
        runner.cpu.writeData(TCNT0, 0);
        currentTime = 0;
        lastTime = currentTime;
        runner.cpu.writeData(runner.portD.portConfig.DDR, 0);
        runner.portD.setPin(2, false);
        while (currentTime - lastTime < 50) {
          runner.cpu.tick();
          runner.cpu.cycles += 64;
          currentTime = runner.cpu.readData(TCNT0) * 4;
        }

        let delay = 30;
        if (data[i] === 0) {
          delay = 30;
        } else if (data[i] === 1) {
          delay = 70;
        }
        runner.cpu.writeData(TCCR0B, CS01 | CS00);
        runner.cpu.writeData(TCNT0, 0);
        currentTime = 0;
        lastTime = currentTime;
        runner.cpu.writeData(runner.portD.portConfig.DDR, 0);
        runner.portD.setPin(2, true);
        while (currentTime - lastTime < delay) {
          runner.cpu.tick();
          runner.cpu.cycles += 64;
          currentTime = runner.cpu.readData(TCNT0) * 4;
        }
      }
      // runner.cpu.writeData(TCCR0B, 0);
      firstFlag = false;
      firstPinState = 0;
      secondPinState = 0;
      lastPinState = 0;
      console.log('end');
    }`

urish, 2023-01-11:
The recommended way to implement it is using `cpu.addClockEvent()` and register a callback to send the next bit each time. This way, the main program keeps running, and your callback is called after a given number of CPU cycles.

xucheng2008, 2023-01-11:
That's great.！It works！

urish, 2023-01-11:
Awesome! What are you building?

xucheng2008, 2023-01-12:
I'm exploring the educational feasibility of AVRJS. So far, so effective. 
Thank you for your contribution. 
Keep trying!

closed 2023-01-12 by xucheng2008

urish, 2023-01-12:
Thanks!

If you end up using it for education, please let us know. We'd love to share the story on our blog!