wokwi/avr8js #148: How does avr8js parse diagram.json into the CPU

Green-hcq, 2023-10-26:
Hi ,
Recently, I tried using wokwi elements and avr8js to create an Arduino simulator, but it doesn't seem very optimistic. The tutorial on avr8js doesn't mention how to verify the positive and negative connections, which makes me confused.

I want to write my own online simulation platform based on two frameworks. can you provide valuable suggestions?

Green-hcq, 2023-10-26:
At present, the most important thing is AVR8js how to verify whether the circuit is connected correctly

urish, 2023-10-26:
Hi Felix, avrj8s only emulates the AVR core of the Arduino. It is not a complete electronics simulator. 

There's an example of integrating avr8js with CircuitJS1, to create a full electronics sim: https://www.falstad.com/circuit/avr8js/ (example source code: https://github.com/MarkMegarry/AVR8js-Falstad).

Another open source example which includes several parts (from wokwi-elements) connected to AVR8js: https://github.com/arcostasi/avr8js-electron-playground.

> At present, the most important thing is AVR8js how to verify whether the circuit is connected correctly

Unfortunately, there is no trivial way to do it. You first have to define what kind of validation you are interested in, and then, based on that, and your requirements, perform either SPICE simulation, some kind of digital simulation (like Wokwi does), write a bunch of rules, or use an LLM such as GPT-4 to determine the correctness of the circuit.

closed 2023-10-26 by urish

Green-hcq, 2023-10-27:
> Hi Felix, avrj8s only emulates the AVR core of the Arduino. It is not a complete electronics simulator.
>
> There's an example of integrating avr8js with CircuitJS1, to create a full electronics sim: [...]

Thank you very much for your suggestion