wokwi/avr8js #158: Two Serial.begin() stops serial output for 328P

ArminJo, 2024-05-30:
It is simple to reproduce, but here is an example: https://wokwi.com/projects/399320984336916481
I use this function to receive modbus requests at 9600 and in between to print short debug messages with 115200.
See https://wokwi.com/projects/399302567914368001 , where  following Serial.begin() are disabled if 
`#define STANDALONE_TEST`
is activated 😞

urish, 2025-02-11:
Thanks for reporting! Have you tested this on a physical device? Does it behave differently?

ArminJo, 2025-02-20:
I just tested it, and
```
Serial.begin(115200);
Serial.begin(115200);
```
as well as 
```
Serial.begin(115200);
IrSender.begin(); // example for any code
Serial.begin(115200);
```
works on Nano hardware.